mjrgh/PinscapePico #7: [bug] Pico Worker[0] not booting into Bootloader mode when attached to board (Mainboard DIY-friendly)

Rappelbox, 2025-01-28:
As the title already describes, I am not possible to force the pico (Worker[0] on DIY-friendly mainbaord) into bootloader mode by pressing the bootsel button when attaching the usb cable.
Thankfully i flashed the firmware before i soldered it to the board. For this reason i know it is functional as it is detected by the GUI.
The other 2 worker picos on the power board can be programmed without issues.

mjrgh, 2025-01-28:
Wow, I'm glad it worked out that you were able to program it before the bug kicked in.

Do you have any idea what the underlying cause is?  Is it that the aux Pico is getting powered from the main Pico, so it's not in RESET at the time you're plugging it in?  If that's it, can you get it into Boot Loader mode by this sequence?:

- Press and hold BOOTSEL on the Worker Pico
- Press and release the RESET button on the main board
- Release BOOTSEL

The RESET button SHOULD do a hard reset (via the RUN pin) on all four Picos, so that was intended to be the way to get any of the Picos into Boot Loader mode if you couldn't do it via software commands.  Once the firmware is loaded onto any of them, you can get them into Boot Loader mode by software command (assuming the software is working properly), but I'd definitely like to be sure you can do a hard Boot Loader reset on any of them as a failsafe against software failures.

mjrgh, 2025-01-28:
By the way, if the RESET button doesn't work, you should hopefully still be able to get any of the Worker Picos into hard reset by manually jumpering the RUN pin to ground.  But I don't want anyone to have to do that - I want to make sure the RESET button works as intended, so if that's not doing it we should definitely track down what's going wrong.

Rappelbox, 2025-01-29:
interesting find for the worker[0]:
bootsel + usb[0]  = no bootloader mode possible
usb[0] + bootsel + RESET button = no bootloader possible
usb[Main] + bootsel + usb[worker] = no bootloader possible, Worker is still powered by main, so this is as expected.
usb[all] + bootsel + RESET button = success! 

This only applies to worker[0], worker[1,2] can be programmed if bootsel + usb[1,2] is plugged directly and main is unplugged

mjrgh, 2025-01-29:
Thanks for trying the various combinations.  

Tell me if you disagree, but it looks to me like everything is working as expected.  Just to make sure we agree on the parameters, let's spell out how BOOTSEL works: BOOTSEL determines what the Pico does **during a power-on reset**, selecting between normal flash startup (BOOTSEL not pressed) and boot loader ROM startup (BOOTSEL pressed).  BOOTSEL has essentially no effect **except** when a power-on reset is in progress.  There are two ways to trigger a power-on reset cycle: (a) power-cycle the Pico; (b) ground the RUN pin. Crucially - this is where the various test cases you've been doing come in - unplugging and plugging the USB cable **doesn't** initiate a power-on reset by itself.  Cycling the USB cable is often listed as shorthand method for getting into BOOTSEL mode, but whenever I mention this, I always try to be careful to qualify it by saying "unplug the USB cable **and all other power sources**", because it's not the USB cable that gets you into BOOTSEL mode, it's the power cycle.  If the Pico is powered by some source other than the USB cable, cycling the USB cable doesn't interrupt system power, therefore doesn't initiate a power-on reset, and therefore can't enter boot loader mode, no matter what's going on with the BOOTSEL button at the time.  By design, the auxiliary (Worker) Picos on these boards are powered from the primary Pico's USB connection, so as long as the primary Pico is plugged into power, the aux Picos are also powered, so their USB connections can come and go without triggering resets.  So the only way to get one of these to reset is via the Reset button, which is wired to ground the RUN pin on each Pico.

Does that make sense?  Are there any cases you've observed that don't agree with this, where something needs to be fixed on the board?

Rappelbox, 2025-01-29:
I agree with you completely on all terms.

I tried those combinations exactly how i described them, maybe i did not explain detailed enough.
As a startup routine exactly and ONLY the mentioned parts are powered, meaning that in case 1 the worker[0] should boot into bootloader mode as there are no other power sources be it pico[main] or secondary power.

mjrgh, 2025-01-29:
Okay, so you're saying there is one case here where we do have an anomaly?  Namely **bootsel + usb[0]**?  I.e., nothing is connected to USB, you press and hold BOOTSEL, plug in Worker 0 USB, release BOOTSEL, it doesn't enter boot loader mode.  Very strange if so... I can't think of anything the board could be doing that's preventing boot loader mode in this case.  I mean, I don't even know how I could deliberately contrive to prevent boot loader mode in that case, since BOOTSEL is so hard-wired internally to the Pico itself.  It seems like the only way that it wouldn't go into boot loader mode is if a POR isn't occurring, and I don't know how you prevent a POR when power is first applied, so this seems to suggest that the Pico is getting power from somewhere.  I don't think there's nearly enough capacitance on the board to be powering it from capacitors.  I'll have to think about it, but I sure don't see any explanation off-hand.

Rappelbox, 2025-01-29:
Correct, for case 1 and 2 bootloader mode is not possible.
I don't know the exact reason. I'll inspect the board tomorrow to check if there is a hardware fault somewhere or a short.

mjrgh, 2025-01-29:
It would definitely be good to figure out what's going on.  Maybe check with a voltmeter when everything is unplugged to see if there's somehow a voltage at the VSYS on the Worker 0 Pico (which is connected to the cathode on D1).  

I guess the only other possibility is that the BOOTSEL button signal is getting lost in this case, but I can't see how that's even possible, since that's completely internal to the Pico - it's not exposed on any external pins.  Even if something were shorted externally, it still shouldn't be able to touch the BOOTSEL wiring.

mjrgh, 2025-01-29:
One other possibility comes to mind: maybe the problem is that the Worker Pico isn't powering up at all from its USB cable.  So it's not that it isn't going into boot loader mode, but rather that it's just not in ANY mode, because it's not working on USB power.

I think we could test for this case as follows: unplug all cables from the board.  Plug in just the Worker Pico USB, **without** pressing BOOTSEL, to let it boot normally under USB power. Check the on-board Pico LED to see if it's blinking briefly every couple of seconds - that's the PWMWorker firmware diagnostic display for "standby mode" before it starts getting I2C instructions from the main Pico.  If the light is blinking, the Pico has power and the PWMWorker firmware is running.  If the light isn't blinking, I think the whole problem is that the Pico just isn't powering up from the USB cable at all, so it can't enter boot loader mode OR normal mode since it's simply not powered.

Rappelbox, 2025-01-29:
Good catch!
I just did a quick test before bedtime. Worker[0] doesn't boot at all when powered alone through usb. Reason needs to be determined.

mjrgh, 2025-01-29:
Great, that should narrow it down a lot.  Makes much more sense than a mysterious power source or something blocking the BOOTSEL wiring.

It seems like the main possibilities are (1) the USB port or USB cable aren't providing power, (2) the main board wiring is shorting the USB power supply to ground so that the Pico isn't getting power, or (3) the RUN pin is being pulled low when main power is off. 

(1) should be easy enough to check by swapping cables and ports with the one you've been using for the main USB in, since that one seems to be working reliably.

(2) is probably worth checking with a voltmeter - test the Pico's 3V3 pin to see if it's at 3.3V when the USB is plugged in.  If it's not, there's probably something going on with the VSYS wiring.  I just double-checked the Powering Pico section in the Pico Datasheet to see if I screwed up the VSYS power input in the design, but it looks right as far as I can see.  So I guess you should check if D1 to make sure it's not installed backwards - if it were, I think it would be shunting VSYS to ground when no power is applied at the primary Pico VBUS, which would prevent the aux Pico from getting power.   The stripe side should be pointing towards the square pad, which is the USB-cable side of the board.

If both of those check out good, then it must be that RUN is getting pulled low when main power is off.   You can check that with a voltmeter too - it should read at 3.3V (roughly) when power is on and the RESET button isn't being pressed.  That connects to three points: the RESET button, through D2, which should be floating as long as you don't press RESET; the collector of Q16, which should be floating as long as the main Pico isn't actively pulling /PERIPHERAL_POWER_ENABLE high (which it can't do as long as its power is off), and the 3V3 line on the main Pico, through the 10K pull-up R32.  So if it's getting pulled low, it's probably because the main Pico's 3V3 is getting pulled low when the power is off.  That could actually be the culprit, if the regulator is actively pulling that line low when it's not powered.  I think it might have been a mistake to include that pull-up - the only thing /AUX_RESET does is tell the secondary Picos to reset by pulling their RUN pins low, and their RUN pins already have their own internal pull-ups, so I don't think there's actually any need for that pull-up to be there at all.  If this turns out to be the problem, I think you can just remove R32 entirely.  I don't think it needs to be there and it might be actively causing this problem.

Rappelbox, 2025-01-30:
Removing R32 did the trick.

Before testing your suggestions i redid my tests and noticed that not only worker[0] can't be forced into bootloader mode, but instead none of the worker picos can be programmed.
In my initial programming i did it without the mainboard being attached so it worked back then.
USB cable was good as it was the same that i used as connection to the PC.

Conclusion: remove R32 from the layout

closed 2025-01-30 by Rappelbox

mjrgh, 2025-01-30:
Great, glad we have an explanation.  I'll delete the resistor from the layout.

mjrgh, 2025-01-31:
Some additional follow-up:  I just checked in a new rev of the boards that reworks the Reset scheme a bit.  The errant pull-up resistor in the old design was there because the RP2040 data sheet recommends an external pull-up on the RUN pin, to reduce the chances of electrical noise on the pin triggering a reset.  I haven't ever noticed that happening with my testing Picos (which don't have external pull-ups), but it seems like a good idea to follow their recommendations.  The problem with the old board was that I was pulling up all of the SEDONARY Pico RUN pins to the PRIMARY Pico's 3V3, which is fine as long as they're all powered, but triggered this bug when the primary Pico was unpowered (hence its 3V3 was at GND, driving the secondary Pico's RUN pin low and forcing it into reset).  So the new design adds back the pull-ups, but it gives each Pico its own isolated pull-up to its own 3V3.  That should accomplish the data sheet recommendation of providing a strong external pull-up, but still work properly regardless of the power status of the other Picos.

The initial simple solution of just removing the problem resistor works because the RP2040 has an internal pull-up on RUN, so you don't absolutely need an external one.  The data sheet wants you to add an external one because the internal one is weak, and they're worried that it's too weak to overcome 100% of noise on the pad.  If it's such a worry, I have to wonder why they didn't just make it strong enough to satisfy themselves, but here we are.  I think I got a little lucky that the design has exactly three aux Picos, because if there were four, I think the bug would manifest again even without the external pull-up.  The original design has all of the RUN pins on the three secondary Picos wired together in parallel, which means that the internal pull-ups are going to form a voltage divider when some of the Picos are powered and some aren't.  If we have two powered and one unpowered, the voltage divider puts the RUN pin at 1.1V, which is just a little above the 3.3V logic-low cutoff level of around 0.7V - not a big margin of safety, but probably enough for reliable operation when you're updating firmware.  It's NOT a problem in normal operation when all of the Picos are powered, since they're all going to be pulling to 3.3V.  But it still made me want to rethink the RESET network.  The simplest thing seemed to be to give each Pico a separate hard Reset switch and a separate NPN switch for the /PERIPHERAL_POWER_ENABLE control, since that isolates the RUN pins and lets each one have its own pull-up to its own 3V3 supply.