mjrgh/PinscapePico #8: Wrong SetPWMWorkerAddr output response

Rappelbox, 2025-01-30:
SetPWMWorkerAddr always responds with:
"Successfully updated device I2C address on D:\ to 0x10" no matter what address is passed.
Also happens when invalid address is passed.

Worker[1,2] not getting detected at 0x31 and 0x32 in log

mjrgh, 2025-01-30:
[SetPWMWorkerAddr-patch.zip](https://github.com/user-attachments/files/18606945/SetPWMWorkerAddr-patch.zip)

D'oh! I somehow deleted a key word in the line that parses the address, in such a way that the code still compiled (but got the address all wrong).   Should be fixed now - patch attached.

Rappelbox, 2025-01-30:
thanks! it works with the new file now as intended.

Important: defender detects a trojan, imo a false positive

closed 2025-01-30 by Rappelbox

mjrgh, 2025-01-31:
> Important: defender detects a trojan, imo a false positive

I assume so - there are no third-party libraries involved, just the Microsoft C++ native libraries.  According to the Web, this is a common false positive with .exe's built with the Microsoft compiler.  If you want to be sure, you can build the program yourself - it shouldn't require the Pico SDK, just Visual Studio with C++, and shouldn't have any outside dependencies outside of the git repos.