mjrgh/PinscapePico #9: ads1115 not being detected/configured

Rappelbox, 2025-01-30:
log: 2025/01/30 21:52:26 Error:   Plunger: no plunger sensor is available; at least one sensor type must be configured

this is the config for plunger and ads1115 sensor:

```
 plunger: {
         source: "ads1115",  // source sensor type; can be omitted if only one plunger sensor is configured
         autoZero: true,          // enable auto-zeroing; only applies to quadrature sensors, ignored otherwise
         autoZeroTime: 5000,      // time in milliseconds for auto-zeroing; plunger must be stationary this long to auto-zero
         powerLaw: 2,             // power-law exponent in brightness-to-distance calculation, only for VCNL4010
	// cannel: 0,
    },
    
    ads1115: {
        i2c: 1,                 // I2C bus 1
        addr: 0x48,             // I2C address, set on the chip via pin inputs, 0x48 to 0x4B
        ready: 11,              // READY pin GPIO
        channel: "AIN0/AIN3",   // plunger input = differential signal AIN0-AIN3
        voltageRange: 2.048,    // voltage range +/- 2.048V from the +1.65V AIN3 reference
    },
```

mjrgh, 2025-01-30:
Are you seeing any errors about the ADS1115 configuration?

Rappelbox, 2025-01-30:
No, this is the only error that shows up in the log but also no further information about the ads1115 being initialized so something like that.

mjrgh, 2025-01-30:
Hmm... hopefully that just means there aren't any errors, but let's make sure that it actually IS getting configured.  Make sure "config" is included in the logging.filter:

```
logging: {
    filter: "debug error warning info config",
}
```

With that, you should see some success messages for the ADS1115 in the startup reports.

If you have a CDC or UART serial port configured, you can also open a PuTTY window on the device and see if the ads1115 command is enabled - `ads1115 -s` should show the configuration and sampling statistics.

Rappelbox, 2025-01-30:
I noticed the configuration is different in the starter.json config file, maybe the difference calculation -2/2v and and configuration is not incorporated in the firmware? Just a guess.

mjrgh, 2025-01-30:
The firmware lets you tell it how the inputs are physically configured, either single-ended or differential.  Starter.json is just one possibility, showing the simpler single-ended setup.  The expansion board has a voltage splitter going to AIN3 that provides a 1.65V reference voltage, so that you can configure it with the AIN0/AIN3 differential setup and use the smaller amplifier setting - you basically get an extra bit of resolution by doing that, by cutting the voltage range from +/- 4V (relative to 0V GND reference) to +/- 2V (relative to the 1.65V reference on AIN3).  So the way you've got it in the snippet above is the right way for the expansion board.

Rappelbox, 2025-01-31:
putty result:
```
> ads1115 -s
ads1115: no such command (type 'help' for a list of available commands)
```

typing help doens't show any sign of ads1115 sensor

```
Command summary:

   74hc165           74HC165 input shift register chip diagnostics
   devtest           special development testing functions
   flash             display the byte contents of flash pages
   fsck              show flash data storage area status
   gamepad           HID gamepad emulation options
   help              list all commands, or show a command's usage details ("help <command>")
   hidstats          show HID status and statistics
   i2c0              I2C bus interface test functions
   i2c1              I2C bus interface test functions
   kb                HID keyboard emulation options
   lis3dh            show LIS3DH accelerometer status
   logger            set and view logger options
   loopstats         show main loop timing statistics
   ls                list files in the flash data storage area
   mem               show memory usage statistics
   nudge             Nudge device information
   out               set output port levels (simulate DOF commands)
   pindev            Open Pinball Controller HID device options
   reboot            hardware-reset the Pico
   rm                delete files from the flash storage area
   statusled         Status LED testing functions
   tvon              test TV ON functions
   version           show firmware and hardware information
   worker            Worker Pico control options
   xinput            XBox controller emulation options
```
looks to me like the ads1115 sensor is not being initialized by any routine so it can be configured but never gets called

Rappelbox, 2025-01-31:
bus scan on i2c1 shows that the address is correct

```
> i2c1 --bus-scan
Starting bus scan on I2C1
2025/01/31 11:30:44 Info:    I2C1 bus scan: found device at 0x18
2025/01/31 11:30:44 Info:    I2C1 bus scan: found device at 0x30
2025/01/31 11:30:44 Info:    I2C1 bus scan: found device at 0x31
2025/01/31 11:30:44 Info:    I2C1 bus scan: found device at 0x32
2025/01/31 11:30:44 Info:    I2C1 bus scan: found device at 0x48
2025/01/31 11:30:45 Info:    I2C1 bus scan completed

> i2c1 -s
Current device: 0
State:          Ready

Device 0: DS1307, addr 0x68
  TX/RX started: 1
  TX completed:  113
  RX completed:  0
  TX timed out:  0
  TX aborted:    1

Device 1: WorkerPico, addr 0x30
  TX/RX started: 1
  TX completed:  1
  RX completed:  0
  TX timed out:  0
  TX aborted:    0

Device 2: WorkerPico, addr 0x31
  TX/RX started: 1
  TX completed:  1
  RX completed:  0
  TX timed out:  0
  TX aborted:    0

Device 3: WorkerPico, addr 0x32
  TX/RX started: 1
  TX completed:  1
  RX completed:  0
  TX timed out:  0
  TX aborted:    0

Device 4: LIS3DH, addr 0x18
  TX/RX started: 66924
  TX completed:  0
  RX completed:  66924
  TX timed out:  0
  TX aborted:    0
>

```

mjrgh, 2025-01-31:
Yeah, it looks like the device is responding to I2C pings, so if you're not seeing any mention of it whatsoever in the log, I'd say the ads1115 config section just isn't getting scanned.

Did you run an error check in the config editor?  My first guess would be that there's just some JSON problem where the ads1115 block isn't in the right place or something, given that there are no errors.  If the ads1115 configurator were seeing the JSON section at all, it should acknowledge that in the log - and given that the device isn't getting configured, there should be logged error messages explaining why.  The complete absence of any sign of ads1115 in the log suggests that the section isn't getting scanned at all.  If you don't have any luck spotting the problem, post the full file and I'll see if I can spot it.

Rappelbox, 2025-01-31:
For testing purpose I used your original file and made only 2 changes:

1. corrected address for lis3dh to "0x18" instead of "0x4c"
2. removed commentation for plunger and ads1115

```
// Pinscape Pico - sample configuration for the Pinscape Pico DIY-Friendly Expansion Board
// Copyright 2024, 2025 Michael J Roberts / BSD-3-Clause license / NO WARRANTY
//
// This is an example of the typical hardware configuration for the
// Pinscape Pico DIY-Friendly Expansion Board.  This reflects the fixed
// wiring between the Pico and the peripherals built into the board.

{
    // Unit identification
    id:
    {
        // Every configuration should have a unit number, to identify the
        // device to DOF.  Assign unit #1 to the first Pinscape Pico in your
        // system, 2 to the second, and so on.  (Don't count KL25Z Pinscape
        // units here.  DOF treats the KL25Z and Pico versions as separate
        // devices, so the Picos get their own numbering starting at 1.)
        // The numbering should always start at 1 and go up sequentially
        // from there, because it'll make things easier to set up with DOF.
        unitNum: 1,

        // The unit name is just for display purposes, to help you tell
        // your Pinscape Pico units apart if you have more than one.  The
        // name will be displayed when the Config Tool or other tools list
        // the available devices on-screen.  The name can be whatever you
        // want, but it's best to keep it fairly short, since it's used to
        // label small UI elements, such as buttons.
        unitName: "Expansion Board",

        // Emulated LedWiz ID.  This is for use by an LedWiz API emulator
        // on the PC host, which provides a virtual LedWiz interface for
        // legacy LedWiz-aware applications that can't access the Pinscape
        // Pico through either DOF or the Pinscape native interfaces.  An
        // LedWiz emulator on the PC can make the Pico appear as an LedWiz
        // to such older applications, allowing those apps to program
        // feedback effects through the Pinscape output ports.  This sets
        // the LedWiz ID, 1 to 16, for the virtual LedWiz.  An LedWiz unit
        // only has 32 ports, so if the Pico is configured with more than
        // 32 logical outputs, the LedWiz emulator will create multiple
        // virtual LedWiz units, one for each group of 32 Pico ports.
        // The additional units will be numbered sequentially starting
        // with the ID assigned here.
        //
        // Note that the Pico doesn't directly emulate the LedWiz USB
        // protocol.  Instead, the emulation is done on the host side.
        // On Windows, this is implemented through an open-source
        // replacement for the LEDWIZ.DLL that comes with the genuine
        // LedWiz.  The replacement DLL is available here:
        //
        //   http://mjrnet.org/pinscape/dll-updates.html#LedWiz.dll
        //
        // If you also have a real LedWiz device in your system, or any
        // other device that emulates an LedWiz, you can change this
        // setting as needed to avoid conflicts with the other devices.
        // Each LedWiz in your system (real or emulated) must be given
        // a unique unit number.  Most real LedWiz units are hard-coded
        // to use unit #1, so you can avoid conflicts with those by
        // changing the Pinscape Pico to use unit #2.  If you don't
        // have any conflicting devices, though, it's better to leave
        // this at the default setting of unit #1, because some of the
        // older LedWiz-aware software doesn't know how to access any
        // units except #1.
        // 
        // To disable LedWiz emulation entirely for this Pico, set
        // ledWizUnitNum to 0.
[... 1349 lines cut ...]
    // the USB port.  Release the button.  The Pico should now appear
    // as a virtual USB thumb drive on the Windows desktop, which
    // you can use for .UF2 firmware installation and for address
    // updates via the SetPWMWorkerAddr program.)
    workerPico: [
        // main board auxiliary Pico
        { i2c: 1, addr: 0x30, pwmFreq: 20000 },

        // power board Pico #1
        { i2c: 1, addr: 0x31, pwmFreq: 20000 },

        // power board Pico #2
        { i2c: 1, addr: 0x32, pwmFreq: 20000 },
    ],

    // Real-time clock chip.  This keeps track of the wall-clock time
    // and calendar date when the Pico is powered off.  A coin cell
    // battery can be installed on the board to keep the RTC chip
    // running even when USB is unpowered or disconnected.
    ds1307: {
        i2c: 1,
    },
}

```

mjrgh, 2025-01-31:
That looks completely fine - I'm stumped.  It's working on my test Pico with an ADS1115, so I don't think I broke something recently in the firmware.  So you're just not seeing any  mention of ADS1115 at all in the log?  Do you have the log buffer big enough that you're certain you're seeing messages all the way back to the initial boot?

Rappelbox, 2025-01-31:
i think so, as said, nothing changed except the 2 things.

when i access trough putty terminal i should be able to see and call plunger or ads1115, right?

mjrgh, 2025-01-31:
> when i access trough putty terminal i should be able to see and call plunger or ads1115, right?

The ads1115 command is added when the ads1115 is configured, so if it's failing or skipping configuration, the command is never added.  So the absence of the command means it's not getting configured.

Do you have the SDK set up so you can build the firmware?  I think the next step is to add some instrumentation to see why the ads1115 configurator either isn't getting invoked or isn't doing its thing.

Rappelbox, 2025-01-31:
Not yet, but I'll try tomorrow morning.

mjrgh, 2025-01-31:
Great, let me know how that goes.  I don't think it should be very hard to set up from my 2.1 SDK snapshot, but I haven't set up a brand new machine in a while to verify the full procedure.  If it does turn out to be too much work to set up, I can send you instrumented firmware builds, but turnaround will be a lot quicker if you're able to mess around the source code directly to try new things out iteratively.

One really quick thing you might try first, though, is cutting out the whole ads1115: { } section and moving it to the top of the file, within the open brace of the top-level object.  Just to make sure there's not some context from a prior object that's blocking it for some reason.  I really doubt that, because it passes the schema check - but the editor and firmware use the same JSON parser, so maybe there's a shared bug that's somehow blocking out that whole object at the JSON parse level.  So maybe worth trying as a sanity check.

mjrgh, 2025-01-31:
> corrected address for lis3dh to "0x18" instead of "0x4c"

Oops - holdover from the old design where I had an MC3416 instead of LIS3DH.

Regards,
Mike

Rappelbox, 2025-02-01:
pico sdk installed

the position of the ads1115{ } section makes no difference, I tested various locations, none of them make a difference in terms of detection.

what i noticed is that if I only define the ads1115 I get a
ERROR "Plunger: no plunger sensor is available; at least one sensor type must be configured"

if i additionally add the pico_adc { }

"Warning: Plunger: multiple sensors are available, Potentiomter(Pico on-board ADC channel 0) selected; to select a different sensor, specify plunger.source in the configuration"

If i specify the ads1115 via 
source: "ads1115",
in the plunger section 
I get again the

"Error:   Plunger: sensor source "ads1115" is unknown, or sensor not configured"

For me it narrows it a bit down that the ads1115 { } is being recognized but the configuration is not recognized by the plunger part or that the ads11115 doesn't finish its initialization routine?!

I left the JSON check ON to avoid stupid mistakes on my end

Rappelbox, 2025-02-01:
log with debug, bufsize16384:

```
 output buffer size 8192, history size 256
0000/01/01 00:00:00 Config:  Button 0: Pushbutton created, action Key(0xE1), source 74HC165[0] chip 0 port A, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 1: Pushbutton created, action Key(0xE5), source 74HC165[0] chip 0 port B, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 2: Pushbutton created, action Key(0xE3), source 74HC165[0] chip 0 port C, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 3: Pushbutton created, action Key(0xE6), source 74HC165[0] chip 0 port D, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 4: Pushbutton created, action Key(0xE0), source 74HC165[0] chip 0 port H, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 5: Pushbutton created, action Key(0xE4), source 74HC165[0] chip 0 port G, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 6: Pushbutton created, action Key(0x1E), source 74HC165[0] chip 0 port F, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 7: Pushbutton created, action Key(0x29), source 74HC165[0] chip 0 port E, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 8: Pushbutton created, action Key(0x20), source 74HC165[0] chip 1 port A, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 9: Pushbutton created, action Key(0x21), source 74HC165[0] chip 1 port B, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 10: Pushbutton created, action Key(0x22), source 74HC165[0] chip 1 port C, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 11: Pushbutton created, action Key(0x23), source 74HC165[0] chip 1 port D, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 12: Pushbutton created, action Key(0x28), source 74HC165[0] chip 1 port H, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 13: Pushbutton created, action Key(0x27), source 74HC165[0] chip 1 port G, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 14: Pushbutton created, action Key(0x26), source 74HC165[0] chip 1 port F, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 15: Pushbutton created, action Key(0x25), source 74HC165[0] chip 1 port E, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 16: Pushbutton created, action Key(0x24), source 74HC165[0] chip 2 port A, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 17: Pushbutton created, action Key(0x4D), source 74HC165[0] chip 2 port B, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 18: Pushbutton created, action Key(0x1F), source 74HC165[0] chip 2 port C, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 19: Pushbutton created, action Key(0x17), source 74HC165[0] chip 2 port D, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 20: Pushbutton created, action Key(0x4A), source 74HC165[0] chip 2 port H, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 21: Pushbutton created, action Key(0x2E), source 74HC165[0] chip 2 port G, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 22: Pushbutton created, action Key(0x2D), source 74HC165[0] chip 2 port F, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 23: Pushbutton created, action GamepadButton(1), source 74HC165[0] chip 2 port E, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 24: Pushbutton created, action GamepadButton(2), source 74HC165[0] chip 3 port A, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 25: Pushbutton created, action GamepadButton(3), source 74HC165[0] chip 3 port B, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 26: Pushbutton created, action GamepadButton(4), source 74HC165[0] chip 3 port C, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 27: Pushbutton created, action GamepadButton(5), source 74HC165[0] chip 3 port D, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 28: Pushbutton created, action GamepadButton(6), source 74HC165[0] chip 3 port H, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 29: Pushbutton created, action GamepadButton(7), source 74HC165[0] chip 3 port G, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 30: Pushbutton created, action GamepadButton(8), source 74HC165[0] chip 3 port F, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Button 31: Pushbutton created, action GamepadButton(9), source 74HC165[0] chip 3 port E, shift(mask 00, bits 00)
0000/01/01 00:00:00 Config:  Output[1]: device GP8 [PWM], time limit 0, reduced power 0, cooling 0
0000/01/01 00:00:00 Config:  Output[2]: device GP9 [PWM], time limit 0, reduced power 0, cooling 0
0000/01/01 00:00:00 Config:  Output[3]: device GP14 [PWM], time limit 0, reduced power 0, cooling 0
0000/01/01 00:00:00 Config:  Output[4]: device GP15 [PWM], time limit 0, reduced power 0, cooling 0
0000/01/01 00:00:00 Config:  Output[5]: device WorkerPico[0] OUT16, time limit 0, reduced power 0, cooling 0
0000/01/01 00:00:00 Config:  Output[6]: device WorkerPico[0] OUT17, time limit 0, reduced power 0, cooling 0
0000/01/01 00:00:00 Config:  Output[7]: device WorkerPico[0] OUT18, time limit 0, reduced power 0, cooling 0
0000/01/01 00:00:00 Config:  Output[8]: device WorkerPico[0] OUT19, time limit 0, reduced power 0, cooling 0
0000/01/01 00:00:00 Config:  Output[9]: device WorkerPico[0] OUT22, time limit 0, reduced power 0, cooling 0
[... cut ...]
0000/01/01 00:00:00 Config:  Output[76]: device WorkerPico[2] OUT5, time limit 0, reduced power 0, cooling 0
0000/01/01 00:00:00 Config:  RGB Status LED configured on R=GP3, G=GP1, B=GP2
0000/01/01 00:00:00 Config:  Nudge device: LIS3DH selected as accelerometer input
0000/01/01 00:00:00 Debug:   File "lis3dh.acc" opened for read, 133 us
0000/01/01 00:00:00 Config:  Nudge input configured on LIS3DH; nudge x = device +X, y = +Y, z = +Z
0000/01/01 00:00:00 Error:   Plunger: sensor source "ads1115" is unknown, or sensor not configured
0000/01/01 00:00:00 Info:    Kb LED 01

```

mjrgh, 2025-02-01:
output buffer size 8192, history size 256
0000/01/01 00:00:00 Config:  Button 0: Pushbutton created, action Key(0xE1), source 74HC165[0] chip 0 port A, shift(mask 00,

Ah, okay, if that's where the log you've been seeing all this time starts, that explains why we're not seeing any ADS1115 messages.  I should have made sure we were talking about the same thing when I was saying "the full log" earlier.  You need to see these messages to be sure you're seeing all the way back to the big bang:

```
===========================================
PinscapePico v0.1.2, build 202501311617

0000/01/01 00:00:00 Info:    Flash JEDEC ID recognized as Winbond W25Q16JV-IQ/JQ, capacity 2097152 bytes
0000/01/01 00:00:00 Info:    SFDP basic parameters table found, capacity entry found [4:7]{ ff ff ff 00 }, 2097152 bytes
```

So let's try increasing the buffer size further - hopefully 32K will do it.  I guess with so many buttons and outputs on the board, the log buffer fills up pretty quickly just with those.

Alternatively, if you have a UART adapter, that's a great way to see everything, since it'll  stay connected across Pico resets.

Let's see what the log is really saying at the start, and maybe we won't need to add any instrumentation.

Rappelbox, 2025-02-02:
got it working, seems it is very "picky" with interpreting values...

1. if only a single channel is defined the sensor is being initialized
`channel: 0, `
2. dual channel channel also initializes if the channel entry is
`channel: ["0/3"], `
the quotation marks for multiple values MUST be in square brackets, else the error pops up:
`Error:   Plunger: sensor source "ads1115" is unknown, or sensor not configured`

BUT: only the first channel is being read, or the debug output is not "precise", or i interpret it wrong ;)

log for single channel:
```
2025/02/02 10:11:21 Info:    Flash JEDEC ID recognized as Winbond W25Q16JV-IQ/JQ, capacity 2097152 bytes
2025/02/02 10:11:21 Info:    SFDP basic parameters table found, capacity entry found [4:7]{ ff ff ff 00 }, 2097152 bytes
2025/02/02 10:11:21 Info:    Flash file system mounted; 266240 bytes allocated to 4 files
2025/02/02 10:11:21 Debug:   File "config.json" opened for read, 33251 us
2025/02/02 10:11:21 Config:  Config file loaded OK
2025/02/02 10:11:21 Config:  UART console enabled, output buffer size 8192, history size 256
2025/02/02 10:11:21 Config:  UART serial port TX configured on DMA channel 0
2025/02/02 10:11:21 Config:  Expansion board peripheral power enable configured on GP5, active low, power-off time 250 ms, wait time 250 ms
2025/02/02 10:11:21 Info:    Resetting peripherals by turning off power for 250 ms
2025/02/02 10:11:21 Info:    Peripheral power restored; waiting 250 ms for devices to cycle
2025/02/02 10:11:21 Config:  I2C0 set for on-demand initialization
2025/02/02 10:11:21 Info:    I2C1 bus reset OK, 0 clocks sent
2025/02/02 10:11:21 Config:  I2C1 configured on SDA=GP6, SCL=GP7; speed: 400000; DMA channels: TX 1, RX 2
2025/02/02 10:11:21 Config:  workerPico[0] configured on I2C1 addr 0x30
2025/02/02 10:11:21 Config:  workerPico[1] configured on I2C1 addr 0x31
2025/02/02 10:11:21 Config:  workerPico[2] configured on I2C1 addr 0x32
2025/02/02 10:11:21 Config:  WorkerPico[0], I2C1 address 0x30: device initialization OK, WHOAMI=0x24 (OK), Ver 1
2025/02/02 10:11:21 Config:  WorkerPico[1], I2C1 address 0x31: device initialization OK, WHOAMI=0x24 (OK), Ver 1
2025/02/02 10:11:21 Config:  WorkerPico[2], I2C1 address 0x32: device initialization OK, WHOAMI=0x24 (OK), Ver 1
2025/02/02 10:11:21 Config:  74HC165[0] initialized, CLK=GP19, QH=GP18, SH/LD=GP20 (LD is active low); DMA ch 3; PIO 0.0@27, clock divider=10.417
2025/02/02 10:11:21 Info:    LIS3DH: WHO_AM_I (reg 0x0F)=0x33 (OK)
2025/02/02 10:11:21 Config:  LIS3DH device initialization OK; interrupt GP22, dynamic range +/- 2g
2025/02/02 10:11:21 Config:  ads1115[0] device initialization OK
2025/02/02 10:11:21 Config:  ADS1115[0] configured on I2C1 addr 0x48, 860 samples/second (DR=111), full-scale voltage range +/- 2.048V (PGA=010)
2025/02/02 10:11:21 Config:  Open Pinball Device HID configured
2025/02/02 10:11:21 Config:  USB CDC console enabled, output buffer size 32767, history size 256
2025/02/02 10:11:21 Config:  RGB Status LED configured on R=GP3, G=GP1, B=GP2
2025/02/02 10:11:21 Config:  Nudge device: LIS3DH selected as accelerometer input
2025/02/02 10:11:21 Debug:   File "lis3dh.acc" opened for read, 130 us
2025/02/02 10:11:21 Config:  Nudge input configured on LIS3DH; nudge x = device +X, y = +Y, z = +Z
2025/02/02 10:11:21 Config:  Plunger: Potentiomter(ADS1115 ADC [0] channel 0) sensor source selected
2025/02/02 10:11:21 Debug:   File "PotPlunger.cal" not found in open for read
2025/02/02 10:11:21 Info:    Flash struct 'PotPlunger.cal' not found (flash storage status code -1); using defaults
2025/02/02 10:11:21 Config:  No saved plunger calibration data found; using defaults
2025/02/02 10:11:21 Info:    Kb LED 01
2025/02/02 10:11:26 Debug:   File "config.json" opened for read, 36395 us
```

log for dual channel:
```
2025/02/02 10:09:42 Info:    Flash JEDEC ID recognized as Winbond W25Q16JV-IQ/JQ, capacity 2097152 bytes
2025/02/02 10:09:42 Info:    SFDP basic parameters table found, capacity entry found [4:7]{ ff ff ff 00 }, 2097152 bytes
2025/02/02 10:09:42 Info:    Flash file system mounted; 266240 bytes allocated to 4 files
2025/02/02 10:09:42 Debug:   File "config.json" opened for read, 33269 us
2025/02/02 10:09:42 Config:  Config file loaded OK
2025/02/02 10:09:42 Config:  UART console enabled, output buffer size 8192, history size 256
2025/02/02 10:09:42 Config:  UART serial port TX configured on DMA channel 0
2025/02/02 10:09:42 Config:  Expansion board peripheral power enable configured on GP5, active low, power-off time 250 ms, wait time 250 ms
2025/02/02 10:09:42 Info:    Resetting peripherals by turning off power for 250 ms
2025/02/02 10:09:42 Info:    Peripheral power restored; waiting 250 ms for devices to cycle
2025/02/02 10:09:42 Config:  I2C0 set for on-demand initialization
2025/02/02 10:09:42 Info:    I2C1 bus reset OK, 0 clocks sent
2025/02/02 10:09:42 Config:  I2C1 configured on SDA=GP6, SCL=GP7; speed: 400000; DMA channels: TX 1, RX 2
2025/02/02 10:09:42 Config:  workerPico[0] configured on I2C1 addr 0x30
2025/02/02 10:09:42 Config:  workerPico[1] configured on I2C1 addr 0x31
2025/02/02 10:09:42 Config:  workerPico[2] configured on I2C1 addr 0x32
2025/02/02 10:09:42 Config:  WorkerPico[0], I2C1 address 0x30: device initialization OK, WHOAMI=0x24 (OK), Ver 1
2025/02/02 10:09:42 Config:  WorkerPico[1], I2C1 address 0x31: device initialization OK, WHOAMI=0x24 (OK), Ver 1
[... cut ...]
2025/02/02 10:09:42 Config:  Nudge input configured on LIS3DH; nudge x = device +X, y = +Y, z = +Z
2025/02/02 10:09:42 Config:  Plunger: Potentiomter(ADS1115 ADC [0] channel 0) sensor source selected
2025/02/02 10:09:42 Debug:   File "PotPlunger.cal" not found in open for read
2025/02/02 10:09:42 Info:    Flash struct 'PotPlunger.cal' not found (flash storage status code -1); using defaults
2025/02/02 10:09:42 Config:  No saved plunger calibration data found; using defaults
2025/02/02 10:09:42 Info:    Kb LED 01
2025/02/02 10:09:47 Debug:   File "config.json" opened for read, 36419 us
```

mjrgh, 2025-02-02:
Great, glad we found a way past the mysterious lack of messages.

Re the "picky" syntax: it would be helpful if you could do two things to follow up:

* For each case that you think it's getting wrong, send me the channel: setting that it's getting wrong, and what you expected the result to be for that setting
* It would also help a lot if you could proofread the ads1115[].channel section in the config ref and let me know if you spot anything wrong, misleading, confusing, or incomplete there

Rappelbox, 2025-02-04:
test was just done:

The syntax is not correct.

All combinations of your helper work when placed inside square brackets []
So, these were the ones I tested:
channel: ["0"],
channel: ["AIN0"],
channel: [0],
channel: ["0/3"],
channel: ["AIN0/AIN3"],

Rappelbox, 2025-02-04:
fixed with commit: https://github.com/mjrgh/PinscapePico/commit/90fb40d612cc68fad004f3f7260c6f4e732adcf2

closed 2025-02-04 by Rappelbox