mjrgh/PinscapePico #10: Very first I2C communication fails

michaelkay, 2025-01-31:
I have set up different Pico configurations and I2C interfaces. I have found that the first I2C transaction always fails with a PICO_ERROR_GENERIC. It doesn't affect the steady state run but initialization can show false error messages. Test setup: LIS3DH setup as the only I2C device (nudge device). Buttons and LED definitions don't matter. Log will show 
_Error:   LIS3DH device initialization failed; interrupt GP18, dynamic range +/- 2g_

I tested and confirmed by adding the following to LIS3DH.cpp around line 165.

		Log(LOG_DEBUG, "LIS3DH: Read WHO_AM_I (reg 0x0F)=0x%02x 0x%02x\n", 
			i2c_write_timeout_us(i2c, i2cAddr, buf, _countof(buf), true, 4000), 
			i2c_read_timeout_us(i2c, i2cAddr, &whoAmI, 1, false, 1000));

This may be a SDK error. I am using the 2.1SDK setup from the PinScape documentation. The other accelerometers don't seem to be affected because there is a read loop in the init routines that can get past this error case. But, this may affect other I2C startup routines from the other interface devices. 

Also it may be the cause of
[https://github.com/mjrgh/PinscapePico/issues/9](https://github.com/mjrgh/PinscapePico/issues/9)

Rappelbox, 2025-01-31:
What i2c-address did you use?
I have the lis3dh working here in combination with the diy-expansion boards.

mjrgh, 2025-01-31:
I haven't seen any I2C-related problems in the 2.1 SDK, so I doubt that's it, especially if other I2C devices are working for you.  So I'd probably take the error at face value and assume that the hardware isn't responding.

If you've enabled the CDC or UART terminal, you can open a terminal session and do a full scan of what's on the bus via `i2c0 --bus-scan` (or `i2c1` as needed).  That should tell you if you've got the wrong address or the device just isn't responding at all.