mjrgh/PinscapePico #22: Request: Gamepad Hat Support

richt314, 2025-04-04:
I have a (somewhat unique, admittedly) application that would benefit from having a d-pad / POV Hat handled by the virtual gamepad device rather than cursor keys in the keyboard emulation, generic gamepad buttons, or d-pad in xinput.  Any thoughts about adding Hat support?

I could probably hack at the source to implement this as a one-off for myself, but figured I'd check and see if there were any plans or interest in bringing it to the wider audience.

mjrgh, 2025-04-04:
I could add that to the base gamepad object pretty easily.  I'd want to think about it a little first to make sure there wouldn't be a downside, but off-hand I can't see any obvious problem with it.  I think the ideal would be to make the gamepad usage list itself configurable, rather than using the current fixed layout, but there are some definite downsides to that, mostly that some of the PC-side software (like DirectInput) is coded to assume that a given device always has the same HID report layout (a safe assumption for a physical gamepad where the button layout is fixed in its physical form).

You'd just want a hat control that's essentially a set of four additional on/off buttons, I assume?  )As opposed to another set of analog X/Y axes.)

richt314, 2025-04-05:
Yes, I'm looking for a digital hat, four directional on/off buttons.

I'm using my Pinscape Pico with an embedded Linux machine, which can only see the first 16 buttons of the HID gamepad (plus the analog axes). These are mapped to gamepad buttons (shoulder buttons, start, select, A/B/X/Y, etc) by the Linux input driver. So by tweaking the Pinscape Pico configuration, I can tweak my buttons to match the gamepad buttons I want, but I can't map anything to directional controls since the Input driver is looking for an HID hat control which doesn't exist. 

If Pinscape Pico had a hat control, I wouldn't need to go out of my way to remap extra gamepad "buttons" to directional controls, leaving all the mapping on the Pinscape side, and I would avoid spending four of the 16 buttons the system could see on my cabinet's d-pad.

mjrgh, 2025-04-05:
I took a quick look at the HID usage tables, and they're pretty vague about how hat switches are supposed to be mapped.  Actually, they're COMPLETELY vague; they just assign Hat Switch a usage code (Generic Desktop Page, usage 0x39) and apparently leave it up to applications to define it.

A little more web searching says that the common practice is to map the hat switch to a field with 9 values, so a 4-bit field with logical min/max 0/8 or 1/9, with the integer values representing the compass points counterclockwise, starting at north, with the minimum value being neutral.  So with a logical range 0..8, it would be 0=neutral, 1=N, 2=NW, 3=W, 4=SW, 5=S, 6=SE, 7=E, 8=NE.  

To map this to four physical button inputs on the Pinscape side, the gamepad layer would take the four button states and generate the corresponding compass point value.  Note that this means that you don't actually have four buttons per se in the final HID input to the PC, since almost half of the 16 possible button states can't be represented in the 9-state HID input.  For example, pressing the N and S button simultaneously can't be represented as a value state.

This would all be pretty straightforward to implement, but given how specific your use case is, AND given that none of the Windows pinball programs even know what a hat switch is, I'm thinking that I should probably let you implement a custom test run first, if you're up for that, to make sure that this proposed mapping actually works for you.  It sounds from your description that what you really want is a set of four additional buttons, and the hat switch doesn't really accomplish that fully, given the limitation that it can't represent all 16 states of four orthogonal buttons.  So I'm a little hesitant to implement it because I'm worried that you'll try it out and find that it doesn't do what you wanted after all - if it doesn't, I think it'll end up being dead code, since I don't think anyone on Windows will have any use for it either given the limitations of the PC pinball programs.

richt314, 2025-04-05:
I see what you mean, the implementation of a hat switch in the HID descriptor from four directional buttons is interesting:
https://pit.uriba.org/uriba/road-to-fcc-part-ii-hid-shenanigans-a-tale-of-a-hat-switch/

A four-way (or 8 way with two buttons at once) directional pad (where pressing opposing up/down or left/right switches at the same time is physically impossible) is exactly what I'm trying to accomplish, for reasons of navigating menus and other normal d-pad tasks. So I see that I would have to map a set of four gpio inputs to a 0-7 range (plus 8 for the no-input case), assuming that page is correct. 

I'm happy to try to hack something together. I can almost guarantee it won't be pretty, but maybe enough of a proof-of-concept to show that it's useful.

mjrgh, 2025-04-05:
Yeah, that looks like yet another set of conventions - this one is 0-7 with 8 as the Null.  I'm curious how flexible the host-side drivers are about interpreting all of these different conventions.

I think I can put together an initial rough implementation pretty quickly.  That should give you a framework to start with, and you can modify it as needed to get it working with your host.  I suspect your host is going to be less flexible than Windows about the report format, but I should be able to at least verify that it works with the Windows joystick clients, so hopefully it won't need more than some small adjustments.

richt314, 2025-04-05:
If you're willing to do that, that would be great. I can play around with a couple different configurations to see how Windows and this particular embedded Linux system respond to them. I can try out the gamepads and joysticks that I have with hat controls as well, try to figure out what the prevailing convention is among them and how the different operating systems respond. 

I appreciate all the help and insight. This started with just interfacing an analog plunger (where my cabinet used to have a digital button instead). Out of curiosity I tried running a flipper button through my Pico as well, and noticed that I could squeeze out 1-2 frames of latency (at 60 Hz) compared to the original interface. Now I'm trying to get everything switched over. Not that I need super fast latency on a d-pad, but it would be nice to have everything unified on a single controller.

mjrgh, 2025-04-05:
I put together a test version for you to try - it's on branch [hat-switch](/mjrgh/PinscapePico/tree/hat-switch).  I figured I'd provide it in source form so you could tinker with the details of the HID mapping if that turns out to be necessary.  That's in firmware/gamepad.cpp, search for "hat" to find the affected code sections.  I did a little quick test with the Windows joy.cpl tester, and it seems good there, but we'll have to see if your Linux setup accepts it.

I'm glad to hear you've had good results with respect to latency so far.  I tried to make sure button latencies were as low as possible, but it's always good to find out it's working in the real world as well!

richt314, 2025-04-06:
Great, thank you!  I'll give it a try when I have a chance and will let you know how it goes.

richt314, 2025-04-08:
Still a little testing to do, but after I got the project to build, the hat switch is showing in Windows and Linux!

Only one big issue with functionality, the rotation "around the axis" was inverted in both the Windows "Game Controllers" dialog and in the embedded Linux system (essentially, left and right were swapped).  I did a little more looking, and the sources I see online seem to suggest a clockwise rotation from "North".  So I inverted the mapping in the firmware source, and all seems to be well.

In Gamepad.cpp, starting at line 305:
`        static const uint8_t hatSwitchMap[] {
            // HID    Angle  Dir   R L D U   Hex
            0,     // NULL   NULL  0 0 0 0   0x00
            1,     // 0      N     0 0 0 1   0x01
            5,     // 180    S     0 0 1 0   0x02
            0,     // NULL   NULL  0 0 1 1   0x03
            7,     // 90     W     0 1 0 0   0x04
            8,     // 45     NW    0 1 0 1   0x05
            6,     // 135    SW    0 1 1 0   0x06
            0,     // NULL   NULL  0 1 1 1   0x07
            3,     // 270    E     1 0 0 0   0x08
            2,     // 315    NE    1 0 0 1   0x09
            4,     // 225    SE    1 0 1 0   0x0A
            0,     // NULL   NULL  1 0 1 1   0x0B
            0,     // NULL   NULL  1 1 0 0   0x0C
            0,     // NULL   NULL  1 1 0 1   0x0D
            0,     // NULL   NULL  1 1 1 0   0x0E
            0,     // NULL   NULL  1 1 1 1   0x0F
        };`

Interesting that it seemed to be working when you tested - did you test "left" and "right" directions?  I am fairly certain this isn't a case of me mapping my buttons backwards, the correct buttons were showing in the "Buttons" tab of the config tool.

Other than that, the GUIConfigTool complains about the hatSwitch property in the config (though ignoring the error works fine).

A few notes about the build process, not sure if these are expected issues or not, but they tripped me up when following the "How-to-build" document:

- I used your pico-sdk-2.1.0.  I had to edit PICO_INSTALL_PATH in version.ini to match where I had unzipped the SDK before pico-env.cmd would work.  Took me a bit to figure out since it wasn't in "How-to-build", but I guess if I had looked in README.md of the SDK I would have figured that out sooner.
- I had to copy "pick_sdk_import.cmake" from pico-sdk to the Firmware folder before cmake would work.  I see a recommended procedure elsewhere of creating a symlink to the file within the SDK.
- Visual Studio kept failing to build the rest of the project, particularly GUIConfigTool.  It came down to ConfigDocParser failing to create GUIConfigTool\Generated\ConfigSchema.h.  After some playing, I figured out that I had to create the (empty) "Generated" folder first, then the build process worked fine. 

Like I said, I'll do some more playing around, but initial indications are this seems to be a worthwhile feature to me, and working well.  I'll let you know how it works out for my use case as I have more time to play around.

mjrgh, 2025-04-08:
Glad to hear it's (mostly) working so far!

> (essentially, left and right were swapped)

I just double-checked it, and it is indeed reversed on my system as well.  I must have just seen what I was looking for, that the left and right buttons were activating the horizontal axis, and ignored that they were backwards.  I'm glad you found that Windows and Linux were consistent on how the coordinate system is interpreted.

Thanks for the notes on the build hurdles - I'll fix/document those...

> I had to edit PICO_INSTALL_PATH

Yeah, that's mentioned in the README.md in the SDK download itself.  I'll add a note to How-to-build that there are some extra steps required, with a pointer there.

> I had to copy "pick_sdk_import.cmake" from pico-sdk to the Firmware folder

I guess I did that so long ago that I forgot it was a required step.  I think I'll just add that to the source tree since it seems like Raspberry Pi intends for it to be used that way.

> I figured out that I had to create the (empty) "Generated" folder

I'll add that to the source tree as well.