mjrgh/PinscapePico #25: Key Chords in new Version

StifflersM0M, 2025-04-07:
Hi!

I just tried the new Key CHords instead of a Macro, but it seems it is not working.
i added

> 	 {
             source: { type: "gpio", gp: 13 },        	  // button is wired to GPIO port GP13
             action: { type: "key", key: "Ctrl+X" }, 
         },

But within the buttons view the GPIO is marked as "none". Also when pushing the button the key are not displayed.
I tried to cut a text in Notepad, but does not worked.

mjrgh, 2025-04-07:
What does the log say about it?

StifflersM0M, 2025-04-07:
Sorry. After looking into the log, i saw, that i was still on the old firmware. I just copied the FIles into my folder and forgot to flash the Firmware onto the board.

Thanks!

closed 2025-04-07 by StifflersM0M