mjrgh/PinscapePico #36: Error opening file "config.json" for reading: CRC failure (stored 2144df1c, computed a67fc7e9)

mkotek, 2025-05-16:
Given the situation with new KL25Z Rev. I being useless, I have tried to switch to Pico. The only thing I had on hand is RPi [RP2040-Zero (Pico clone)](https://pl.aliexpress.com/item/1005003823256706.html), which was recognized fine and firmware upload has been successful. When trying to update the configuration, it seems to run fine, but then the log file shows error: Error opening file "config.json" for reading: CRC failure (stored 2144df1c, computed a67fc7e9) and the configuration is not stored.
I have tried to use the command line tool and all seems to go fine, yet the moment I try to reload the configuration using GUI, the error shows again.

First of all, is Zero even supported? If so, why am I getting file system corrupted information?

mjrgh, 2025-05-16:
I haven't tried it on that clone.  The AliExpress page you linked to isn't working, so I can't tell if you have a genuine clone or a clone of a clone (Waveshare seems to claim credit as the "original" RP2040-Zero clone, with their warning about counterfeits of their clones).  From the CRC error, it sounds like the flash-erase-write that the firmware has to do to store the config data isn't working properly, either not storing anything or storing garbled data.  If you want to do some debugging, you can connect a PuTTY terminal to the Pinscape command console (see **Serial Ports** in the JSON config reference), and take a look at what's in flash.  Use **ls** to get the file list; that'll tell you the starting location (StrmAddr) for the config file (config.json).  Than you can use **flash _addr_** to display portions of the data stored there.  The config file is stored contiguously starting at the stream address, so you can look through it to see if it looks the data is stored at all, and if so, if it looks corrupted.  **ls** will also check the CRC and report on validity.

mkotek, 2025-05-16:
Thank you for the immediate response. How do I define the virtual COM port? I have this in the definition:
        usb: {
            logging: true,       // enable logging on this port
            console: {
                enable: true,    // enable the interactive command line on this port
                bufSize: 2048,   // buffer size for console output
            },
        },
    },

mjrgh, 2025-05-16:
> How do I define the virtual COM port? 

See **Serial Ports** in the doc.

mjrgh, 2025-05-16:
I should add: if the JSON isn't getting stored, you probably won't be able to get the serial port working, since the firmware won't be able to load the JSON that isn't getting stored.  I don't think you'll be able to use that board if that's the case - either it's a broken specimen, or that particular clone is just incompatible after all.

mkotek, 2025-05-16:
The latter might be the case, the AliExpress link should work (it works for me, but it was just an example, not the actual seller I got the board from).

mjrgh, 2025-05-16:
If you want to get really into debugging it, you can create a custom firmware build that enables the serial ports without a config file. You'll have to install the Pico SDK and build environment and build the firmware from source, but that shouldn't be too difficult if you use my build environment setup guide (see **How-to-build.md** on the main github page).  With a custom build, you can set the "factory" settings to anything you like by populating the **factoryConfig[]** string defined at the top of **Config::Load()** in firmware/config.cpp.  Just put in the same serialPort section that you'd normally put in the JSON file to enable the ports.

mjrgh, 2025-05-16:
One more data point you can look at is the boot log. When you boot the device, the first couple of lines in the log should have info on the flash chip detected, including ID and size. Do you see anything there?

mkotek, 2025-05-16:
Do you mean this:
Info:    Flash JEDEC ID: Mfg ID 68, Dev ID 10, Capacity ID 15

mjrgh, 2025-05-16:
Exactly - that's part of it, at least. There might also be an "SFDP basic parameters table" line, if the flash chip provides that (it's a quasi industry standard that some but not all chips implement).  At any rate, capacity ID 15 is the code for 2MB (1 << 0x15), so that much at least is fine. The manufacturer/device ID indicate that it's a different chip from the one that Raspberry Pi uses, so maybe that's the issue. But that shouldn't be a problem in principle, since the flash chip obviously has to be compatible enough that the boot ROM can successfully transfer firmware into it.  The Pinscape software invokes the same ROM code to transfer config data, so at that level it seems like it ought to be working.  I think the you'll probably have to get the console working so you can look at the flash contents if you want to investigate further.

mkotek, 2025-05-16:
Not sure it is worth continuing. I just had to try having useless KL25Z now. I am going to get the supported Pico to continue the tests and possibly an old working KL25Z to finish my cab, but thank you for the support.

mjrgh, 2025-05-16:
Yeah, I agree - I'm curious what the problem is, since it might be something simple to fix once diagnosed, but it'll be a lot less work to just get a regular Pico, and hopefully that'll just work well out of the box for you.

mkotek, 2025-05-17:
Is there any difference between getting Pico with micro USB port or USB C?