mjrgh/PinscapePico #47: Pinscape PICO DIY Main/Power odd output results

BrainstormStudios, 2025-08-05:
HI all. I have built many projects in my time as well as arcade repair. I opted to do the DIY Friendly board set Main and Power. I did 2 sets and have sunk a good bit into them so I REALLY want to make this work. With my main board alone, I have good button input and results using the configuration template that appeared to be for this boards. For reference these are from the 20250203 revision. All appears to be good from the assembly standpoint but looking for some direction to getting the output side working as intended in case I may have missed something. What would cause some outputs to trigger when testing via the config tool and the rest no response. Note on each set of boards that were built, each pair has a different set of outputs that work and don't work. Any guidance is welcome as I REALLY want to make this work before saying it's a bad design and moving to a resiny made finished product. I design boards and write software so I understand things can go wrong on these types of projects.

BrainstormStudios, 2025-08-05:
I have made a new post. Thanks for the advisement. Haven't used GitHub for support before.

mjrgh, 2025-08-05:
Great, thanks.

closed 2025-08-05 by mjrgh