mjrgh/PinscapePico #56: Support for MMA845X

Bandicoot37, 2025-08-29:
Hello 
First Thanks a lot for this PinscapePico
My KLZ25 dead after my PSU burnt ^^
I've build a setup with a veroboard and all I need for me ( Plunger , IRremote and Nudge ) work like a charm , so many thanks ;)
But for nudge  i've make a mistake when i bought my accelerometer , i've bought a MMA845X insteed of MMA8451Q
Do you think in future you'll add this or not?

mjrgh, 2025-08-29:
From what I can tell, there's no such thing as a "MMA845X" chip as such; it's just a generic term for the MMA854* family of chips (the "X" means "wildcard").  Do you know the specific version of the chip you have?  There might be markings on the chip that you can use to figure out which it is.  The Pinscape code is specifically for the MMA8541Q, but it might work on the other chips in the series, since the manufacturers often reuse the same register in related chips.  So you might give it a try and see if you can it working as is.  If that doesn't work, see if you can identify the specific chip you have and see if there's a data sheet available.  We can see what kind of changes would be required.  If the differences are minor enough it might be something I could add.

Bandicoot37, 2025-08-29:
yes mine is MMA8542

Bandicoot37, 2025-08-29:
OK go it work ^^
i use @shermand100  config and change 
 mma8451q: {
        i2c: 0,
        addr: 0x1D,
	interrupt: 3,
        gRange: 2,
     },
to
 mma8451q: {
        i2c: 0,
        addr: 0x1C,
	interrupt: 3,
        gRange: 2,
     },

^^

So just change addr: 0x1D to 0x1C and now it's working , thanks