mjrgh/PinscapePico #57: IR send problem

Bandicoot37, 2025-08-29:
Hello
i try to send ircode with ConfigTool
i use this command
ConfigTool --ir-send 05.00.0059
Everything works with the Guiconfigtool but not with the command line
But ConfigTool --ir-learn have no problem
( sorry for two issue at the same time ^^ )
thanks

mjrgh, 2025-08-29:
Can you provide any details about what's going wrong?

Bandicoot37, 2025-08-31:
So
https://github.com/user-attachments/assets/2cc52a24-03d0-4291-ab32-60a8895f3e98
This is learn by CMD
https://github.com/user-attachments/assets/5044d110-8305-4464-88ba-0093b3e0f2c3
This is faulty send by CMD
https://github.com/user-attachments/assets/619223db-cb65-4ae4-80e8-5e93e6786d81
This is working with GUI

Bandicoot37, 2025-10-24:
So many thanks works like a charm ;) ;) ;) big kisss , SDK version 2.1 and 2.2 works for IR , let you closed this issue ;) ;)

mjrgh, 2025-10-24:
Great - thanks for testing it out!

closed 2025-10-24 by mjrgh