mjrgh/PinscapePico #69: output stops working with pulseMode feature

winxi01, 2025-09-14:
Hi, I'm now setting up the contactor outputs using sharegroups and I would like to make use of the pulseMode feature for the flipper outputs. It works quite well and it seems to be possible to replicate the on- and off-clonks quite well by adjusting tOn and tOff accordingly.
However, it happens that the output doesn't work anymore after the flipper buttons are pressed in quick succession. The ouput then stays on and the Pico has to be resetted to make it work again.
I could reproduce this with two simple outputs:
```
{
   device: { type: "workerPico", unit: 0, port: 22 },
   shareGroup: "test",
   noisy: true,
   name: "cbr",
},
{
   device: { type: "shareGroup", group: "test", pulseMode: {tOn: 30, tOff: 22, }, },
   name: "test",
},
```

mjrgh, 2025-09-14:
I gave it a quick try, and I couldn't trigger the problem, so there must be something more to it.  Are you able to reproduce it without DOF Involved, just firing the ports from the config tool?  How about if you change the physical output to a GPIO instead of a Worker Pico?  And how fast exactly is "quick succession"?

winxi01, 2025-09-14:
> Are you able to reproduce it without DOF Involved, just firing the ports from the config tool? 

Couldn't reproduce it with the config tool. It's difficult to test with the config tool, because with the sliders it's not possible to go directly from 0 to 255 and I don't know how Pinscape handles the values in between with shared groups and when pulseMode is active. For this and other test scenarios it would be helpful to have an additional button in the config tool for each output which allows to toggle the corresponding output between 0 and 255.

However, I defined a button as source and with this I was able to reproduce it without DOF being involved. For the test, I had three ouputs defined:
```
outputs: [        
   // Output #1
   {
      device: { type: "workerPico", unit: 0, port: 10 },
      shareGroup: "cbl",
   },
	
   // Output #2
   {
      device: { type: "workerPico", unit: 0, port: 9 },
      source: "port('Flipper Left')"
   },

   // Output #3
   {
      device: { type: "shareGroup", group: "cbl", pulseMode: {tOn: 30, tOff: 22, }, },
      name: "Flipper Left",
      source: "button('left flipper', 255, 0)",
   },	
],
```
I've done a [short video](https://drive.google.com/file/d/17wXj5n2q0RT6F2hSSglZCrvhjZkbFzeb/view?usp=drive_link) where you can see the results of the test. In the DOF Config Tool, I've assigned just `P:003` to "Flipper Left". The yellow lamp on the left is wired to `device: { type: "workerPico", unit: 0, port: 9 }` (`Output #2`) and shows directly the value of `Output #3` and the green lamp is wired to `device: { type: "workerPico", unit: 0, port: 10 }` (`Output #1`) and shows the pulses processed by `pulseMode`. The triggered button is the button with name `'left flipper'`.
In the end, it can be seen that `Output #1` (green lamp) continues to stay on.

> How about if you change the physical output to a GPIO instead of a Worker Pico?

That's a good hint. However, in my device with the DIY-friendly main board, all devices and LEDs are driven by the Worker Pico. This means, that I've to re-wire things. Will be able to test this in the nex days.

> And how fast exactly is "quick succession"?

Unfortunately, I'm not sure how to measure and quantify this. But maybe you get an idea with the video above.
 
> Also, when it stops working, are there any other symptoms? For example, does the whole Pico crash and become unresponsive, or is just the one port that stops working?

The Pico is still working and responsive, just the one port stops working (can also be seen in the video).

winxi01, 2025-09-15:
Did a bit more testing and I think I found a reliable way to reproduce:

- The output window tester shows the port being stuck.
- It can also be reproduced with a regular gpio (monitored in the config tool, as you suggested).
- The port seems to get stuck when an on-puls is triggered before an off-puls has been finished. For testing this, you can set `pulseMode.tOff` to a high value, e.g. 1000. This way, I could reproduce it even with the slider in the config tool.

mjrgh, 2025-09-16:
Reproduced, and I think I found the problem and fixed it.  Test build attached.  Try it out when you get a chance and let me know if this fixes it for you.

[PulseSharedOutputFix.zip](https://github.com/user-attachments/files/22372904/PulseSharedOutputFix.zip)

winxi01, 2025-09-17:
> Try it out when you get a chance and let me know if this fixes it for you.

It works as expected now, couldn't reproduce a port being stuck with the patched version. Thank you very much for the quick fix!